kbwbe/A2plus #342: Can not move part anymore.

Perlovka, 2020-01-21:
Hi, I've just updated from 0.4.29 to latest versions and "Move" button stopped working. Transform dialog is opened instead. 0.4.39 - same behavior.

I'm using FreeCAD 0.18.4

kbwbe, 2020-01-22:
The "Move" function has been replaced by the std. transform tool for more precision. So the parts can be moved around using this.

Perlovka, 2020-01-22:
But transform does not allow to move freely with the mouse, not so handy, IMO.
Is it possible to make parts moving just by grabbing with mouse? Or FreeCAD does not allow such a behavior?

kbwbe, 2020-01-22:
Using the transform tool, it is possible to do more things with the mouse than before. You can move and rotate the part directly in the 3D View by mouse. Rotating wasn't possible before. A little backdraw is that you have to close the dialog, when finished.

Perlovka, 2020-01-22:
Well, before I had two possibilities - move part freely with the mouse and transform. Now - only transform. Don't you think it is regression in usability? There are too many clicks now for just to throw a part aside. Select part -> aim on button "move" and click -> aim on that small arrow tip -> move part -> click OK (Esc).

Please consider make things less complicated to use, I wan't to spam this issue anymore.
This was just my opinion about usability regression from my point of view.
Feel free to just close as won't fix, if you are not willing to fix this ;)
Thank you and have a nice day )

luzpaz, 2023-03-06:
@kbwbe verdict ?