kbwbe/A2plus #346: A2 + loses FCFtd parts images and WRLs

JCB37, 2020-01-29:
When you import a part containing an image plane or a wrl, only the solid parts are displayed in the assembly.

See the example below:
[image]
 2020_0006-1.FCStd with imageplane

[image]
 2020_0006-1.FCStd with WRL 

[image]
A2plus assembly after import of 2020_0006-1.FCStd and 2020_0006-2.FCStd

kbwbe, 2020-01-29:
Image planes are not supported. I do not know what a WRL is. Please give a short explanation.

JCB37, 2020-01-31:
I understand that A2+ can't uses image and WRL file directly because they are not solid parts.
But IMO, when they are included in a standard FCStd, I think they are in a "black box" and can be displayed as the FCStd.
Some information about WRL files :
https://whatis.techtarget.com/fileformat/WRL-Virtual-Reality-model
which 
WRL files are very well  import in FreeCAD and do a very good rendering of the pieces , (but they are not solid)
For example, I join the original WRL files (made with KiCAD), the FCstd made with it (you can see the display of it) and the A2+ importation of the FCStd which incorrectly displayed the FCStc included

[toto.zip](https://github.com/kbwbe/A2plus/files/4138338/toto.zip)

easyw, 2020-01-31:
But why you are not simply using the STEP file if your board?

luzpaz, 2021-02-28:
@JCB37 ping

luzpaz, 2023-03-06:
@kbwbe close ?