kbwbe/A2plus #349: Can't edit existing constraints any more

justinclift, 2020-02-04:
Experimenting with A2plus (v0.4.42a), and noticed the "Edit selected constraint" button isn't consistently available when an existing constraint is selected in the Model tab.

eg:

justinclift, 2020-02-04:
[Oops, pressed Enter when switching between windows and accidentally created this issue prior to having all the info ready.  Still collecting info, for filling out this issue.]

kbwbe, 2020-02-04:
Hi,
i tested this but cannot reproduce the effect. The dialog appears always. When it is visible, the toolbar icon is greyed out. When i close the dialog, the toolbar icon gets active again.

For test i compiled a fresh FreeCAD:

OS: Linux Mint 19.2 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.19478 (Git)
Build type: Unknown
Branch: master
Hash: e6a29986eb7caf569b9353e997f310c1c2efbec8
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

kbwbe, 2020-02-04:
I will have a look at this during my work with A2plus next days (also on other machines)

justinclift, 2020-02-05:
No worries. :smile:

It's possible it's something weird with my system, though it would be a bit surprising as it's not a behaviour I've yet seen from any other application.  I'll try with an updated (again) git compile from newer master in the next few days too.  Just in case it was an unfortunate point in time that I compiled from or something. :smile:

kbwbe, 2021-01-17:
Please open a new issue if the problem persists.

closed 2021-01-17 by kbwbe