kbwbe/A2plus #360: [feature] Make TARGET POS/SPIN-ACCURACY configurable via preferences

Raketenphysik, 2020-02-27:
At least for my workflow it was helpful to decrease the accuracy by 0.1. Make up your mind if it makes sense to allow this via preferences.

>
            solverControlData = {
                #Index:(posAccuracy,spinAccuracy,completeSolvingRequired)
                1:(0.1,0.1,True),
                2:(0.01,0.01,True),
                3:(0.001,0.001,False),
                4:(0.0001,0.0001,False),
                5:(0.00001,0.00001,False)
                }

Thanks in advance.

kbwbe, 2020-03-01:
I think reducing the required accuracy to 0.01 is not very dangerous for all configurations. It only determines the error size when the solver definitely gives up. When ever possible, the solver works up to accuracy better then 1e-5.
I will change the setup of the required accuracy to 1e-2.

Just for interest: Is there anything special with your assemblies ? (very big parts ? A lot of subassemblies stacked on each other?)

Raketenphysik, 2020-03-01:
>  Is there anything special with your assemblies ? (very big parts ? A lot of subassemblies stacked on each other?)

21 Parts all in
11 of them are Sub-Assemblys
9 different Parts at all (7 of them are Sub-Assemblys)
31 (!) Constraints

[feature request] statistics button ;-)

There is one more thing that is a bit special about this Assembly. Two of the parts are alignet by two (!) constraints. For the last (second) constraint to be solvable the lenght of both parts have to be equal. This is not the case ... both parts differ in lenght by 0.001 something mm. This is a very special situation where the solver has a hard time due to unsuitable determining factors ...

Rgrds, Mark

closed 2020-03-01 by kbwbe