kbwbe/A2plus #370: Error loading addon on Linux

Qik000, 2020-03-16:
Linux Kubuntu 19.10
FreeCAD: 0.18.3+dfsg1-1

Error contents:
```
/usr/lib/python3/dist-packages/matplotlib/__init__.py:848: MatplotlibDeprecationWarning: 
The backend.qt4 rcparam was deprecated in Matplotlib 2.2 and will be removed in 3.1. In order to force the use of a specific Qt binding, either import that binding first, or set the QT_API environment variable.
  "2.2", name=key, obj_type="rcparam", addendum=addendum)
```

kbwbe, 2020-03-17:
The error concerns mathplotlib. I do not use it with A2plus. So i do not think A2plus has to do something with that error.

luzpaz, 2020-03-18:
Also @Qik000 isn't running latest stable which is 0.18.4

Qik000, 2020-03-22:
> Also @Qik000 isn't running latest stable which is 0.18.4

Yes, i am terribly sorry for reporting it here, it turned out the addon throwing this error was assembly2

closed 2020-03-22 by Qik000