kbwbe/A2plus #375: Group ->  object has no attribute 'Transparency'

Raketenphysik, 2020-04-04:
Toggling transparency leads to ...

`Traceback (most recent call last):
  File "C:\Users\...\AppData\Roaming\FreeCAD\Mod\A2plus\a2p_importpart.py", line 1437, in Activated
    a2plib.setTransparency()
  File "C:\Users\...\AppData\Roaming\FreeCAD\Mod\A2plus\a2plib.py", line 274, in setTransparency
    (obj.Name, obj.ViewObject.Transparency, obj.ViewObject.ShapeColor, None)
AttributeError: 'Gui.ViewProviderDocumentObject' object has no attribute 'Transparency'
Running the Python command 'a2p_ToggleTransparencyCommand' failed, try to resume`

... if parts are grouped (Group). Find a simple example enclosed. Deleting the group solves the issue.

Thanks in advance! Rgrds,
Mark

[A2plus_transparency_issue.zip](https://github.com/kbwbe/A2plus/files/4431833/A2plus_transparency_issue.zip)

_____________________________________
OS: Windows 8.1 (6.3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.20311 (Git)
Build type: Release
Branch: master
Hash: 915e551bbb7e3614bc804f1ae1f65027b5432b9f
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

kbwbe, 2020-04-05:
Should be fixed with V0.4.46b.

Raketenphysik, 2020-04-05:
Yes, runs like a charm. Thanks + Rgrds, Mark

closed 2020-04-05 by kbwbe