kbwbe/A2plus #378: Poor memory management of A2P Solver

jkadofo, 2020-04-09:
Hello,

First of all, thanks to all who have worked hard and contributed to the development of this addon! This is a great addon, and is getting better and better with time.

I was trying out a modified version of the interactive script example from [here](https://wiki.freecadweb.org/A2plus_Workbench#Interactive_Script_Example). Then along the line I realized the memory usage of FreeCAD had gone very high (using Task Manager). 

After inspecting the python script for creating the interactive slider, I realized the large increase in memory was due to the repeated running of the a2p solver. I also tested by repeatedly clicking the "Solve" button on the toolbar and observed the same effect. I believe this is controlled by the code here https://github.com/kbwbe/A2plus/blob/master/a2p_solversystem.py

I run a quick test using a simpler assembly with fewer parts and fewer constraints, which also exhibited the same problem. In this case however the rate of memory increase was much slower.

It would be great if you can review the code to see if there's anything that can be done to solve this. (I remember from my C programming classes some time ago that calling malloc() without deleting the allocated memory later, for example, can result in a similar effect).

I am using A2plus workbench v0.4.46b (latest version at the time of this post). FreeCAD version is 0.18 and running on Windows 7 64-bit. Attached is a zip file of the assembly I was working on.

[Interactive Crank Motion Assembly Testing.zip](https://github.com/kbwbe/A2plus/files/4454095/Interactive.Crank.Motion.Assembly.Testing.zip)

kbwbe, 2020-04-13:
Hi @jkadofo,
i can confirm memory consumption during each step of "SolveSystem". It really matters if one is doing simulations.
Memory management using Python as programming language is not that easy and some memory leaks cannot be completely avoided. Python is keeping a lot of internal lists, which are immortal and their memory is not given back to the system during lifetime of the program. So it is not possible to delete no more used objects from memory directly, as it is possible with "C/C++". Every object, which is not used anymore, is deleted (or sometimes it is not for various reasons), from Pythons garbage collector.
The code can only be optimized in order to reduce memory consumption. (using __slots__, avoiding string concatenation etc). I will have a look at this during next time, where it is possible. But this is a bunch of work and will need a lot of time. And it will be never perfect.

jkadofo, 2020-04-13:
I see. Thanks for the details on the underlying issues. It will be helpful to whoever will want to take it up. I hope we find a way eventually :-)

closed 2020-04-13 by kbwbe