kbwbe/A2plus #380: Naming objects

leoheck, 2020-04-17:
From where the "Add part from an external file" gets the object name?

My issue is that one file is not being named as in the file name.

When I import most (99%) of my Freecad files, A2Plus gets the file name and append a number at the end in the format to name the imported object as in:
```
File: case_bottom.FCStd
Imported object: case_bottom_001
```

But I have one file `case_top.FCStd` that is getting imported as `case_001`. I guess I named it like that when I created this file some time ago. But how can I fix it now so the model that is imported has the same name as the filename? 

All the information about the that Freecad file looks good tho.

[image]

kbwbe, 2020-04-18:
The naming shown within the tree-view is only the label. You can rename it as you want without breaking the assembly. The underlying object name can not be changed, as it is impossible using FreeCAD.

leoheck, 2020-04-18:
Hi @kbwbe, you didn't answer my question. From where the object name came from? How can I fix it?

kbwbe, 2020-04-18:
In file "a2p_importpart.py" (line about 246) a function of a2plib is called to generate the object name:
        partName = a2plib.findUnusedObjectName( importDoc.Label, document=doc )
and a second function to generate the object label:
            partLabel = a2plib.findUnusedObjectLabel( importDoc.Label, document=doc )

You have to modify those functions in a2plib if you want to have another naming scheme.
At moment, the naming is based on the label of the importDoc.

There also exist the property importDoc.FileName. Be aware, if dealing with such file pathes, you can easily get a lot of problems with infunctional naming for objects and problems with os platform dependencies. So you can experiment at your own risk.

leoheck, 2020-04-19:
> At moment, the naming is based on the label of the importDoc.

Good, thank you for explaining this deeply.

:) I just want to understand what is happening. I don't want to change anything. 

What I want to do is to re-create my file or the object or something else to be able to use the right name in that corner case.

closed 2020-10-28 by kbwbe