kbwbe/A2plus #381: Function within constain offests

povi2946, 2020-04-18:
Hi,
I would like to request for a feature. 
When making planeCoincident constrains I am allowed to set an offset. But I am not allowed to set a formula or use numbers from a spreadsheet.
Also, I usually work in metric but sometimes I need to use inches so I just write 4in or so and it automatically converts within when using other workbenches but in this offset field I am not allowed to enter other system units.. which would be nice. 
Thank you

hfiguiere, 2022-06-28:
The code in commit 5fda4cdd43bf358ad9b0f7c2aa4fec84026c5be8 is part of the problem.

From the example above,
user_qty is `1112.0` and  user_unit is  `+/"` which lead to setting the suffix on the `QDoubleSpinbBox` **the second time** to `+/"`

luzpaz, 2022-06-28:
@DanMiel ping

hfiguiere, 2022-06-28:
I have a partial patch that show the proper `mm` value.

But I think instead of using QDoubleSpinBox, it should use one of the FreeCAD SpinBox. Another problem is that typing something does trigger the solver...