kbwbe/A2plus #386: Parts are not being imported in the active Part (yellow)

leoheck, 2020-05-25:
Parts are not being imported in the active Part (yellow)

kbwbe, 2021-01-17:
A2pus is not made for use with Part containers. They provide an own coordinate system, which can mess up everything. Therefore i will not change this.

closed 2021-01-17 by kbwbe

leoheck, 2021-01-17:
Oh, so it is not possible to use part containers? Damm, this is so bad!! I like them because I can move/hide them separately.
What if A2Plus has such a container without a coordinate system?
Or maybe you are saying these containers are useless and I should not use them?

kbwbe, 2021-01-17:
As i understood from previous discussions, you are using A2plus without using constraints. In this case, nothing is preventing you from using part containers. But you have to move your things into and out manually. A2plus will not support an active part container.

leoheck, 2021-01-17:
Ok. But just to make it clear, I am not using constraints because it never worked for me. I tried that but every time I add more than 2 pieces these constraints end up broken or the parts move to random places. I would like to use this extension to its full extent but I never could make it work. Of course, I may be doing something wrong but I don't know how to make it work. This is why I moved away from this complexity.