kbwbe/A2plus #398: Avoid loading hidden objects

leoheck, 2020-10-25:
Hey @kbwbe is it possible to skip loading hidden objects?

I use to import other Freecad Objects inside the design I am working to evaluate if they mate together... Then I save it, to keep these extra parts there, so I can continue working on it for some days.

Then, while I am designing I also use to load the design that is being created into the final assembly to see the work in progress is ok. But then it also loads extra hidden A2Plus loaded designs that were loaded in the step before.

kbwbe, 2020-10-25:
Did you activate the A2p option "do not import invisible shapes" within the Preferences page ?

leoheck, 2020-10-25:
No, I didn't. To be honest I am learning about this just now.
I was thinking that this could be the default behavior. Is there a reason for this checkbox being unchecked by default?

kbwbe, 2020-10-26:
This is not the default behaviour, because people might break their assemblies, if they do not think about the consequences of making objects invisible, which are constrained.
Does everything work for you now when using the mentioned preference ?

leoheck, 2020-10-26:
To be honest I didn't enable it. Just because of this message that says it can potentially break things. 

This is why I want to know what can happen? 
How things can be broken? 

I am pretty much an experienced user on FreeCad I can possibly solve most of the issues I have, but I cannot see the consequences that "break things" on this action may cause.

I want to enable it but I don't want to break my designs in a way that I cannot fix later since these designs are going to be used in my work.

kbwbe, 2020-10-26:
Example for how to break assemblies by setting parts invisible:
You have a main assembly which imports to different sub assemblies.
Sub-assembly1: e.g. 3 Parts: A1, A2, A3
Sub-assembly2: e.g. 3 Parts: B1, B2, B3

- In main assembly import sub-assembly 1 and sub-assembly 2 and do constraints for example from A2 to B2, save the main assembly
- edit sub-ass. 1 and make A2 invisible. Store it. If the preference is active, A2 will not be imported to the main assembly because it is invisible.

When updating the main assembly after that, the part A2 of the sub assembly1 has vanished and the geometry which is referred by the constraints can not be found anymore.  All constraints refering to A2 are broken.

leoheck, 2020-10-26:
Ah, break in the sense that the constraints won't work anymore. Got it. So is it still possible to open the file and remove the constraints, right? If it is right, then I have nothing to worry about since I don't use constraints at all... they are always broken for me. Never got more than 2 workings in a file. Also, it looks pretty complex to use them. It is pretty said I know, but I am using A2PLus just to load FCStd inside another.

kbwbe, 2020-10-27:
So you can use the mentioned preference without risk.

closed 2020-10-28 by kbwbe