kbwbe/A2plus #399: Parts list: references are corrupt

Ooypunk, 2020-11-27:
In my project I have imported a number of other external files, and in these files I generated and filled a #PARTINFO# sheet.
In the project I have a sheet with all the dimensions, materials, etc. In the "part info" sheets I have references to values in that dimensions sheet, and they look something like this: "=<<Master>>#<<Materials>>.C5".
Now, when I generate the Parts List, all the values of these part info sheets are copied into this sheet, but these references look like this: "=&lt;&lt;Master&gt;&gt;#&lt;&lt;Materials&gt;&gt;.C5".

I solved this by editing file a2p_simpleXMLreader.py:
line 39: added: import xml.sax.saxutils as saxutils
line 142: changed into: cellDict[cellAdress] = saxutils.unescape(a2plib.to_str(cellContent))

I didn't create a pull request, because I have no idea if this is correct (if this doesn't create more problems in other places); this is my first time dabbling in Python in general, and this project in particular.

kbwbe, 2020-11-27:
Hi,
can you provide some very small example files for testing ? Then i can have a look at this problem. Thank you.

Ooypunk, 2020-11-28:
[github-issue-339.zip](https://github.com/kbwbe/A2plus/files/5610325/github-issue-339.zip)

See attached ZIP file.

kbwbe, 2020-11-28:
Thank you for the files. I think your solution is simple and does not affect something else in the workbench. I have implemented your changes to the master branch. Please update to version 0.4.47f.

Ooypunk, 2020-11-28:
Thanks for the speedy response!

closed 2020-11-28 by Ooypunk