kbwbe/A2plus #404: A2Plus is importing parts when the father is hidden.

leoheck, 2021-01-12:
Hi @kbwbe 

I have this issue that I have some folders, and inside I have some parts (Yellow object) with other parts inside.
Then by hiding the folder hides the internal parts, but not the parts inside these yellow parts. Then these things are being imported by A2Plus.

[image]

This is what happens. To be able to export on A2Plus, I have to hide all the idents inside the third level of the tree.
The best thing is just to import objects that the parent is visible. Maybe this is the original intent and I am seeing a bug, maybe not?

I am using v 0.4.47e

kbwbe, 2021-01-17:
I have changed this behaviour. Reason was: Yellow Part containers do not propagate changes in visibility to all its (and subsequent) childs.

A2plus now does not import shapes when at least one of the father group or part containers is invisible. But it does not change any visibilities.

Please update A2plus to version V0.4.47h and test it.

closed 2021-01-19 by kbwbe

leoheck, 2021-01-20:
Nice @kbwbe I appreciate your hard work. This is the best external plugin I could find.