kbwbe/A2plus #405: [Feature Request] Configurable Part Info Fields

Wright4TheJob, 2021-01-13:
The Part List and Part Info fields are central to many industrial workflows for tracking parts and materials. Since each person or organization has a slightly different workflow, being able to customize those fields without re-writing or forking A2Plus would let FreeCAD replace existing software even more easily.

I propose moving the PartInfo label list from the a2p_partlistglobal file to the a2plib file for access and modification via the FreeCAD preferences panel. The preferences panel would be updated to include a simple list of labels to add, remove, or modify the Part Info fields. 

If this is of community interest, I will fork and modify as a first prototype.

kbwbe, 2021-01-17:
Yes, i think this would be a good addition to the existing wb. Please do a prototype. I will add this functionality to the workbench.
Please take care not to expand the size of the preferences page of A2plus. It is getting to big. Best way is to split it into several tabs.

luzpaz, 2021-04-16:
@Wright4TheJob any progress on this ?

Wright4TheJob, 2021-04-22:
None yet, sorry. It is still on the to-do list!

FarmingSoul, 2021-09-18:
you can try that way :
```

import os, shutil

### to have the dir of external configuration file
wbPath = os.path.dirname(__file__)
# custom file
InfoKeysFile       = os.path.join( wbPath, 'InfoKeys.py' )
# default file
InfoKeysFileInit   = os.path.join( wbPath, 'InfoKeysInit.py' )

### try to open existing external configuration file of user
try :
    fichier = open(InfoKeysFile, 'r')
    fichier.close()
    import InfoKeys as InfoKeys
### else make the default external configuration file
except :
    shutil.copyfile( InfoKeysFileInit , InfoKeysFile )
    import InfoKeys as InfoKeys
```

and replace all the PartInfo label to InfoKeys.PartInfo

FarmingSoul, 2021-09-19:
i work on a version of editable field in Asm4 (and auto filled)
the configuration will be done from a ui.
I'll come back to you later

Have nice day

FarmingSoul, 2021-09-19:
i am totaly in accord with you :

> The Part List and Part Info fields are central to many industrial workflows for tracking parts and materials. Since each person or organization has a slightly different workflow, being able to [...]