kbwbe/A2plus #413: duplicate parts and constraint

lamba84, 2021-02-11:
Hi,
in an assembly, after few parts are duplicated (CTRL+C, CTRL+V) and few constraint added, it start rise error: "Constraints: inconsistent: Cannot solve System".

Using same parts, it's possible to fully define the assembly if parts are added one at the time using "add part" toolbox, instead of duplicate.

OS: Windows 10
freecad: 0.19 23970 and 23074
A2plus: 0.4.50

Hope this will help someone else and thank you for your work!

kbwbe, 2021-02-16:
Hi, thanks for your feedback.
A2plus does not make any difference between freshly imported parts and duplicated parts (Theoretically). Has there been exactly the same constraints in the same order been applied ?

smog1, 2021-09-24:
Hello,
I bounce back on the subject. I would like to be able to duplicate elements with their constraints to make a network by modifying only one of the parameters. Thanks