kbwbe/A2plus #415: Constraint problems upon update parts.

thekswenson, 2021-02-16:
I have the model of a drawer assembly. Each part is defined in it's own file, and the proportions are defined in a spreadsheet that is externally defined. I used your nice tool to define constraints on the sides, ends, and bottom etc. of the drawer so that it tightly fits together (I actually overlap these parts since I will join them with tough and groove).

I would like to play with the proportions by changing the values in the spreadsheet and clicking the update parts button. When I do this I get many messages like this:

"Constraint planeCoincident_003 is broken. Delete constraint? Otherwise check for wrong linkage."

I have to click "abort" several times and sometimes the drawer is redrawn properly updated. Sometimes there is no update and I have to click "update parts" again. Sometimes dimensions change but parts end up disassembled.

I think this may have something to do with your disclaimer about topological naming but I'm not sure.  I noticed that you have preference setting for "Use experimental topological naming", which I thought was intended to solve this. Anyway, I'm not sure if there is a bug, this is the expected behavior, or I'm misusing the tool.

It appears that I'm not allowed to upload the files from the project because they are the wrong filetype.

thekswenson, 2021-02-16:
I'm using FreeCADDaily 0.19 (2021/02/14 13:21:00) and A2Plus ( 7b7e97d..cc9e010) installed via the addon manager.

kbwbe, 2021-02-16:
I guess you can upload your files in a zipped folder. 

For sure the mentioned problem has to do with FC's topological naming issue.

The preference setting "Use experimental topological naming" is made to improve this. You have to activate the setting before you add your first part to an assembly. But the methods can only reduce the effects and you do not get rid completely of the topological naming issue. Also the design style is very important when creating new parts to avoid problems. When i am using PartDesign, usually i do a lot of features with very small sketches. This helps a lot, but not always. (and only is the mentioned preference is activated)

If you  update several parts by means of a spreadsheet, the risk of corrupting constraints is very high, as there are many parts involved.

thekswenson, 2021-02-19:
[drawer_files.zip](https://github.com/kbwbe/A2plus/files/6008937/drawer_files.zip)

Here are the files specifying the drawer. All the parts are just padded rectangles.

One of the first things I did was click "Use experimental topological naming", so all of the parts should be  named using that method.

The spreadsheet defining the parts is in cabinet.FCStd. This file defined the entire cabinet, for which I didn't include all the parts. These parts are in this zip file in case it's relevant:
[cabinet_files.zip](https://github.com/kbwbe/A2plus/files/6008959/cabinet_files.zip)

pauledd, 2021-07-04:
I have the same problem. Of cause I enabled "Use experiental topology namimg" prior creating anything in Freecad...

Here is a video with two simple cubes made in Part Design:
[YT-Freecad A2plus Problem](https://www.youtube.com/watch?v=-Gb-mYENT54)
The files:
[Test.zip](https://github.com/kbwbe/A2plus/files/6759545/Test.zip)

The weird thing is indeed that if you do "Update Parts" twice and do both "Abort" it "seems" to work but I am not sure if this somehow creates problematic things in the project...