kbwbe/A2plus #418: Feature Request - Multi object import

leoheck, 2021-02-19:
Hi @kbwbe, Currently the A2Plus has 2 ways to import models. The "Add a part" tool, and the "Add a single shape" tool.
Can we have a 3rd tool that allows adding multiples parts from a single file?
Or even better, changing the "Add a single shape" to "Add shapes from a file" then instead of a list, it shows a window with checkboxes where it is possible to select more than pme object at once to be imported?

leoheck, 2021-02-19:
Ah, even better. Both tools could be replaced by this new tool. And the first and default item could be an option to import the  "whole-part"

kbwbe, 2021-02-22:
Thanks for the input, but no. For vast majority of use cases, the two possibilities are enough.

closed 2021-02-22 by kbwbe

leoheck, 2021-02-22:
Dude, you are so closed on your point of view. It is so bad to even try to talk with you. I am going to switch to another kind of assembly workbench since it looks like you don't want any kind of human interaction. Well, thanks for your plugins so far. But I won't use and recommend it anymore.