kbwbe/A2plus #420: Help with a constraint.

leoheck, 2021-02-21:
Hi, can you help me with this? 

I am trying constraints for the first time. They are really hard for me.
I want to centralize these 2 parts in the center. They don't have the same width, and height. the orange one is 0.4mm smaller than the other piece. 
How can I achieve that?

The AxisCoincidente does not have a horizontal and/or vertical offset.

[image]

kbwbe, 2021-02-22:
planesCoincident isn't possible ? There is an offset which can be used.

leoheck, 2021-02-22:
I will try. I was exploiting but couldn't make it happen.

For instance, it is not working just by selecting these 2 planes. Nothing hapens.
[image]

leoheck, 2021-02-22:
This should be the end result.
[image]

kbwbe, 2021-02-22:
You should have a look at some tutorials at YouTube how constraints are working, before asking here such things. This here is not the right place for help requests. Please have a look at the pinned issue at top of the issue list.

A very good tutorial is here: [german video](https://www.youtube.com/watch?v=vDcaFq6IEJM)

This is from A2plus start page here at github: [video no sound](https://www.youtube.com/watch?v=QMxcQ5tssWk)

I think there are also a lot of guys who will help in FC's assembly forum.

closed 2021-02-22 by kbwbe

leoheck, 2021-02-22:
So, you are the developer of this thing, but you don't want to give a hint on how to use this? Got it, thanks anyway.