kbwbe/A2plus #431: Offset a real value

fandres, 2021-05-07:
Is there a possibility to do the "offset" a real value? It is required for Song President

kbwbe, 2021-05-08:
I have difficulties to understand the problem. You already can enter real values to constraints' offsets. If the precision of 2 digits isn't enough, you can enter more precise values directly to the field offset within the constraints' property editor.

Are you using a recent Version of A2plus ? Otherwise you shoud update it.

fandres, 2021-05-08:
[image]
I'm trying to restrict two planes: I need a 2.35mm offset but it does not allow me floating point, huh attempted with "coma", and "point". 
A2+Plus V0.4.5

kbwbe, 2021-05-08:
What is your A2plus version ? It is displayed within the workbench selector widget.

fandres, 2021-05-08:
A2Plus:
[image]
V0.4.53a 

Freecad:
OS: Debian GNU/Linux 10 (buster) (KDE//usr/share/xsessions/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.504.24301 +3597 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 540995a6e8d5db01dfe2597439ec3131ca69b030
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: Spanish/Colombia (es_CO)

kbwbe, 2021-05-08:
FC Version: 2021.504.24301 +3597 (Git) AppImage
Build type: Release
Branch: LinkDaily

RealThunder's FC branch ?
A2plus is tested with FC-0.16 ... FC-0.19.2, but not with the RT-branch. Perhaps this is the reason why offset precision is not working.

fandres, 2021-05-08:
Branch: LinkStage3 + assembly3

Download:
https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11

I also have Freecad 0.19 Evre if it also happens. It is possible that it is due to a point or coma problem

kbwbe, 2021-05-13:
I had a little bit time and tested input of real values to planeCoincident constraint on LinkStage3/Daily
It worked out of the box. Please see following screenshot:

[image]

Did you set the FC Prefences correctly ?
Preferences/General/Units => Number of decimals

[image]

A2plus is sensitive to that value.

fandres, 2021-05-14:
Hello
I have corroborated and with the cube it works 
[image]

The project that was creating does not work.
Eliminated the directory and reinstalled A2plus already worked, I do not know what was wrong. I was alienating a .step and a piece created in part desing workbench 
[image]

You have been very attentive. Thanks

kbwbe, 2021-05-14:
You are welcome.

closed 2021-05-14 by kbwbe