kbwbe/A2plus #444: New feature

Did-FR, 2021-10-14:
Hello,

I wrote a new feature to A2Plus plugin. I called it TechBook. The goal is to create a uniq PDF file with all the TechDraw found in the elements of the assembly. My plugin needs the python librairy PyPDF2.
I also add auto completion of editable fields in the techdraw.

You can find my code source on github : https://github.com/Did-FR/A2P_Upgrade

Regards,
Didier

luzpaz, 2022-04-03:
@Did-FR neat. Can you announce this on the wiki?

luzpaz, 2022-04-03:
@kbwbe can you rename this ticket: `New Feature: TechBook - generate PDF with all TechDraw found in the elements of assembly`

Did-FR, 2022-04-07:
Hello,

I made the announce on the forum and I changed the title. I hope that you will find the feature usefull.

Regards,
Didier

luzpaz, 2022-04-07:
Please post the forum thread link next time, thanks  
https://forum.freecadweb.org/viewtopic.php?t=67843

luzpaz, 2022-04-07:
@Did-FR there is no explanation for users on how to test it?

luzpaz, 2022-04-09:
@Did-FR ping

luzpaz, 2023-02-19:
Hey @WandererFan thought you may dig this