kbwbe/A2plus #458: CD_A2plusupdater.py has xe2 errors on lines 683 and 684

phillHgl, 2022-03-21:
When I manually copy A2plus folder to FreeCAD\Mod folder, launch FreeCAD and select A2plus workbench I get the following error:

Workbench failure
("Non-ASCII character '\\xe2' in file c:\\Program Files\\FreeCAD 0.18\\Mod\\A2plus\\CD_A2plusupdater.py on line 684, but no encoding declared; see http://python.org/dev/peps/pep-263/ for details", ('C:\\Program Files\\FreeCAD 0.18\\Mod\\A2plus\\CD_A2plusupdater.py', 683, 0, None))

I can avoid this error by editing the the file and removing the single quote character from line 683, but then this message is posted on the single quote character in the next line.  Removing both single quote characters resolved these two error messages.  

OS: Windows 10
Word size of OS: 32-bit
Word size of FreeCAD: 32-bit
Version: 0.18.16093 +17 (Git)
Build type: Release
Branch: (HEAD detached at 0.18.1)
Hash: f7dccfaa909e5b9da26bf50c4a22ccca9bb10c40
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)

Additionally after resolving these errors by removing the single quote characters, I also get the following error (which may not be related to this issue???)

Cannot load workbench
A2pluseWorkbench:
PythonCommand::getResource():Method
GetResources() of the Python command object returns a dictionary which holds not only strings

I would like to update to 0.20 or 0.19 of FreeCAD but I have not figured out how to get a OS 32bit compatible build yet.

kbwbe, 2022-03-21:
I uploaded an update which should fix the problem of the missing encoding and the mentioned single quotes.
I will try to find the problem behind GetResources() failure.

luzpaz, 2022-04-03:
What `GetResources()` failures?

luzpaz, 2022-08-28:
Is this issue solved?

luzpaz, 2022-09-20:
@kbwbe ping

luzpaz, 2023-03-06:
@kbwbe ping