kbwbe/A2plus #485: Negative offset crashes FreeCad

ImadSaddik, 2022-08-22:
Hi! 

I am having a problem with the offset property. As seen in the videos that I have attached, whenever I enter a negative offset, FreeCad crashes whenever I try to edit it in the constraint properties window. On the other hand, changing the offset value in the data view panel is fine.

https://user-images.githubusercontent.com/79410781/185947366-aa6448b1-0aa5-4f87-8e60-cf496e2ef653.mp4
https://user-images.githubusercontent.com/79410781/185949159-ebfa41a2-9d0e-42d6-b197-ab26a0fa9ff9.mp4

More info :
[code]
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git) Snap 161
Build type: Release
Branch: tag: 0.20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.2
  * ExplodedAssembly
  * Behave-Dark-Colors 0.1.1
  * Help 1.0.3
  * BOLTSFC
  * Assembly3 0.11.3
  * Assembly4 0.12.3
  * Curves 0.5.6
  * fcgear 1.0.0
  * ProDarkThemePreferencePack 1.0.0
  * A2plus 0.4.56a
[/code]

kbwbe, 2022-08-24:
Hi,
thanks for reporting. A2plus at moment is not able to evaluate formulas in the constraint editor. In your video i see that the "-" sign is placed before "mm". Is that a specific local setting ? A2plus will only accept the "-" sign before the numerical value.

ImadSaddik, 2022-08-24:
Hello, 

Thanks for answering. The "-mm" is another problem. When I specify the offset, as an example "-6 mm" and hit Accept, I get the desired result. Once I reopen the constraint properties window, the string changes automatically to "6 -mm", see the 1st video for more details.

Note: the 2nd video is just showing FreeCad crashing after trying to make modification to the offset value.

https://user-images.githubusercontent.com/79410781/186511852-45054a68-61b3-43a7-b9d5-06029a6c9ffe.mp4
https://user-images.githubusercontent.com/79410781/186511859-92947c63-777d-4017-93b1-98abfcd8ffc8.mp4

kbwbe, 2022-08-26:
This is really ugly. I will investigate why this is happening.

kbwbe, 2022-09-25:
Please update the a2+ workbench and test again. For me, it seems to work now.

ImadSaddik, 2022-09-25:
Thank you, the problem is solved.

https://user-images.githubusercontent.com/79410781/192168284-457b6b12-f8bf-4fe5-9eb6-ca7a75871d01.mp4

closed 2022-09-25 by ImadSaddik