kbwbe/A2plus #505: Conflict finder + Could not move some parts -dialog

kemes, 2022-09-05:
In a situation where the Conflict Finder is needed there might be multiple parts that can't be moved. In this case the "Could not move some parts" dialog will appear once per each unmovable part. This quite often leads to the need of clicking "ok" to close the dialog annoyingly many times.

An improvement could be that:
1) Clicking OK once would be enough
OR
2) A checkbox "Don't show this dialog for the rest" -type of selection.

kemes, 2022-09-05:
There could also be an option in the A2+ preferences no to show the "Could not move some parts" -dialog. In this case the parts could just be highlighted. By default the dialog could be enabled.

luzpaz, 2022-09-20:
@DanMiel :point_up:

DanMiel, 2022-09-20:
This annoying message box has been around forever. Kbwbe may have worked on it because he uploaded some small Interface changes to the on 2022-09-10. I downloaded the latest version, and then created some constraints between non- fixed parts. On rebuilding A2plus gave one error message and highlighted the floating parts. I then fixed all three parts and rebuilt the assembly. There was no error message but there should have been.
I have written on this error message on the forums before but at that time I didn’t know about requesting changes.   
IMHO:
The error is produced if both parts are fixed or both parts are floating. A separate error should be for floating parts and fixed parts.
At present the highlight disappears when you press close. I would like to see the error message list the parts and stay on the screen so the engineer has a chance to rotate the assembly and find/repair all parts.
 luzpaz: I expressed my dissatisfaction with this error message. How can I help.
Dan

luzpaz, 2022-09-21:
@DanMiel thank you for weighing-in. Do you happen to know the thread you commented on in the forum and x-post it here?

DanMiel, 2022-09-21:
This looks like my error. I would swear I wrote one, but I cannot find a post about this issue. I found a couple that mentions wanting better error messages but not focusing on them.

If you want me to look at how this message is produced and try to give suggestions on how to improve it, I can do that, but I feel kbwbe would like me to stay out of his files.
Dan

luzpaz, 2022-09-21:
Hi @kbwbe, please advise how to proceed ?

kbwbe, 2022-09-25:
@Kemes: thanks for reporting.

@DanMiel: During standard solving, this message is absolutely necessary, as many people do not realize, why floating parts are not moved.  I had to answer such questions very often.

@Kemes, @luzpaz, @DanMiel : During searching for conflicting constraints, this message box is not needed. In case of free floating constraints, there is no conflict existing but only some constraints are missing.
So i changed the code so that during running the conflict finder the message is suppressed. During standard solving the message is active again. In each case the floating parts are highlighted. During conflict finder an additional console print is done.
The changes are already merged to master branch.

So i hope this is a workable compromise for all needs.

DanMiel, 2022-09-25:
Kbwbe: Thanks for responding and thanks for the A2plus program. I am adding and trying to clarify my opinion on error messages.
In answer to the reply to me: I think the error messages are necessary and should not be removed. What I, and I believe kemes, was trying to point out is that one message at the end of checks should be all that is needed, not a popup every time an error is encountered. 
At present the popup allows you to delete all constraints or close the popup, this does not allow the person to copy/paste the list of parts, and the parts do not stay highlighted after the box is closed. IMHO: At minimum the popup would be a textbox that would list the parts and stay active while the user works on the model in the background, this would give them time to rotate the model and view the constraints before the information disappears.
Unclear message: Part of your program looks for “unmoved” parts, your message says “not constrained and “over constrained” parts. "Over constrained" is where to many constraints are used to constrain a part: Your program does not report these and doesn’t need to since this is poor practice, not an error. Your program highlights both floating parts and when both parts of a constraint are fixed.  These are two different errors and I think it would be helpful to the user to identify which constraint has which error.
Dan Miel

kbwbe, 2022-09-25:
@DanMiel : Your comment: '''unclear message: Part of your program looks for “unmoved” parts, your message says “not constrained and “over constrained” parts. "Over constrained" is where to many constraints are used to constrain a part''

That is not written in the message. Perhaps my english is not good enough.
Text of message is:

"" The highlighted parts were not moved. They are not constrained (also over constraint chains) to a fixed part.""

I think is is a complete other meaning as you are describing above. I tried to describe "free floating parts in space".

DanMiel, 2022-09-25:
Kbwbe:  I have attached some pictures. “Three constraints” have three constraints attaching the two parts. These are enough to constrain the parts together. The ‘Overconstraint”  has seven constraints holding the two together, which to me the part is “over constrained” since there are more constraints than needed.  
In “Not fixed” neither part is fixed, parts do not move and your message shows. In “fixed” both parts are fixed but there is no message. I feel that both parts being fixed is an error unless the constraint is suppressed.  I have some programs that check the fixed property of both parts in a constraint. If both parts are fixed, I report this as an error.
For me, I would like to know about the floating parts, but I don’t necessarily worry about them because I float them on purpose. 
When two parts are constrained and both are fixed, to me this is an error.  I thought in older versions of A2plus they were reported with the floating parts.
The unmoved comment was because in a2p_soversystem.py you have on line 445 a sub called “def detectUnmovedparts” which adds unmoved parts to a list that you use to highlight later on. This is a list that I think could be printed to a text file so a user can view the parts.
I hope this helps.
Dan 
I was trying to think of a less confusing error message below, I don’t think I succeeded.
 “The highlighted parts need to be constraint to parts which are eventually linked to a fixed part.” 
[image]
[image]
[image]
[image]

DanMiel, 2022-09-27:
I upgraded the error reporting on my error checker (which is the green check mark in A2plus). I like several thinks about this report:
The last column tells me exactly what I need to look for, clicking on the constraint name shows me where the constraint connects, I can attach the ends of the constraint to another surface using the attach button, I can find edge and vertex constraints using the find w Label and lastly it stays open until I close it. 
I don't like the size; it is too big.
Kbwbe: I'm not saying to get this fancy because when this pops up it hides a lot, but it would be nice if the messages didn't lock the screens.
[image]

luzpaz, 2022-10-29:
bumping

DanMiel, 2022-10-31:
luzpaz: I have entered a PR [https://github.com/kbwbe/A2plus/pull/535](url). This will help clarify the error messages in my files starting with CD_. It does not change anything in kbwbe's files.

DanMiel, 2022-11-06:
luzpaz bumped this so I'll respond again.
I built an assembly in a2plus and added some errors to it. When I run the a2plus “Solve Constraints” I get an error dialog with “Constraints inconsistent. Cannot solve system. Please run the conflict finder tool!” So running the “Conflict finder tool” should find the bad constraints, it does but the only option that is given is to delete the constraint and as I procced, a dialog pops up for each conflict. To find the conflicting constraints I need to write down the names and the part they belong to, then search through the tree for these. The larger the file the larger the search task.
I realized my Constraint checker and kbwbe’s Conflict finder are similar tools. Referring to the attached picture kbwbe’s three dialogs is shown across the top. My dialog is on the bottom and only shows at the end of the check. Looking at the “Problem” column it has located two other constraints with other problems which a user may or may not want to fix (the floating parts may not have been in the assembly when I ran Kbwbe’s checker). Highlighting a constraint’s name highlights the two connecting features of a constraint (easy to find). There are other tools to help find or fix a constraint and the dialog does not lock up the screen.
IMO, even though this is more a more complex tool than simply asking to delete the constraint, it is a more useful tool to diagnose problems.
Dan
[image]