kbwbe/A2plus #510: Bug when modifying constraints Objects in Python

thebruuu, 2022-09-21:
Hello

Here's the scenario i am facing
So First.
I have a Group containing 3 imported Parts. Let's call it [GROUP A]
Those parts have constraints between themselves 

I select all the parts and constraints in the group, 
Then I create a copy of the selected elements (copy and paste or "Duplicate Selection" command in the Edit Menu, the behaviour is the same)

The copied parts have wrong constraints : the constraints point to the original elements in [GROUP A]
To try to have the constraint correcly repluged to the right parts I tried to run this code: 

```
obj1=FreeCADGui.Selection.getSelection()[0]
obj2=FreeCADGui.Selection.getSelection()[1]
obj1.Object1=obj1.ParentTreeObject.Name
obj1.Object2=obj2.Name
```

where first I select the constraint and then I select the 2nd part it should refer to, as the first part is directly obtained from the object tree ...  and then I run the code from the console

Unfortunately this code does not work as expected: 
Yes, I am able to replug the copied constraint correctly to the parts
BUT randomly, some of the original constraints in [GROUP A] are modified too

Any explanation will be welcome, on why the code does not work as expected.
Thank You

luzpaz, 2023-02-19:
Please add your full [About](https://wiki.freecad.org/About) info. Also add an example file (zip the FCStd file and drag it in to the github textbox).

luzpaz, 2023-03-06:
Closing as no response from OP