kbwbe/A2plus #543: A2plus: Cannot Suppress the Constraint in the data field. Suppressed field is grayed out

luzpaz, 2022-11-16:
This was originally posted to https://github.com/FreeCAD/FreeCAD/issues/7810 by @DanMiel. I've opened a ticket here instead and closing the upstream one. 

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30858 (Git)
Build type: Release
Branch: master
Hash: 7a5019b2625a148acb23fb915df48ffa69636fe9
Python 3.8.13, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60b
  * Assembly4 0.12.4
  * BIM 2021.12.0
  * Curves 0.5.11
  * fasteners 0.4.16
  * Manipulator 1.4.9
  * Moveparts 2022.10.28
  * QuickMeasure 2022.10.28
```

### Problem description

After constraints are made between part in A2plus the constraint can normally be suppressed in the data field in the Combo View. The field is a True/False but is grayed out.
Dan 

### Anything else?

[image]

kbwbe, 2022-11-16:
The property "Suppressed" depends on the visibility of the selected constraint.
- Select the constraint
- switch visibility by pressing the space bar
- the property changes state

This behavior exists since a very long time. So there is no new issue.

luzpaz, 2022-11-17:
So i guess the feature succinctly speaking is to make it possible to enable/disable visibility via the data view in the combo view? (not just spacebar)

DanMiel, 2022-11-18:
luzpaz, you are correct. I can see why having it gray when suppressed would help identify suppressed constraints in the tree but I didn't connect a suppressed constraint with visibility or the spacebar.