kbwbe/A2plus #606: Issue importing step files wrt:colors

timiskander, 2023-08-24:
I was having a pretty consistent issue trying to import a step file (PCB model generated by Altium).
The error was coming from TopoMapper.createTopoNames(). Specifically trying to access DiffuseColor (exception thrown saying object has no attibute "DiffuseColor". I ended up working around it by adding
if hasattr(ob.ViewObject,"ShapeColor"):
around the "manage colors of faces" block.
[a2p_import_color.patch](https://github.com/kbwbe/A2plus/files/12430143/a2p_import_color.patch)
I have very little insight into the inner workings of freecad, nevermind the a2plus code base, but the patch is here for your consideration.
cheers and thanks for providing this tool!
/Tim

luzpaz, 2023-11-12:
@kbwbe can you review the patch ?

jhn98032, 2024-05-22:
The problem with that patch is that everything is gray when importing color step files. My fix #629 fixes the crash but without transforming the step file to gray.