kbwbe/A2plus #608: Part import window is empty

nmchristov, 2023-10-15:
Part import window is empty, no files or folders are listed in the Downloads or Documents forlder. 
version 0.4.60
OS Mac 12.6

luzpaz, 2023-11-12:
more details please. 
Full [About](https://wiki.freecad.org/About) info also missing

AlbertEmil, 2024-07-24:
I just encountered the same issue and posted it on the FreeCAD Forum before checking the GitHub repo. I provided a detailed description and system information in my post there:

https://forum.freecad.org/viewtopic.php?t=89372

AlbertEmil, 2024-07-25:
Seems like this issue is related to a missing parameter, which is not set automatically upon installing the A2plus workbench. Defining the parameter manually as shown in [this post in the FreeCAD Forum](https://forum.freecad.org/viewtopic.php?p=773101#p773009) solved the issue for me.