kbwbe/A2plus #609: Generated parts list contain HTML codes

glassondesignstudio, 2023-11-04:
I have been using the Parts List features for a little while and recently noticed that part information created very recently contains HTML &apos; codes whereas older files don't contain those codes. The screen shot below shows the issue. The two entries at the top of the parts list were created around six months ago while the part information for bottom two entries were created about six weeks ago. In terms of FreeCAD versions I think the last two were created around about the time 0.21.0 or 0.21.1 came out (but don't quote me on that).

[image]

I would like to submit a patch for your consideration. The patch uses a regular expression to remove  the unwanted HTML codes. Please note that regular expressions are not my strong suit and I used the solution that I found [on stackoverflow](https://stackoverflow.com/questions/9662346/python-code-to-remove-html-tags-from-a-string).

[a2p_Bom_py.txt](https://github.com/kbwbe/A2plus/files/13256050/a2p_Bom_py.txt)

glassondesignstudio, 2023-11-04:
Forgot to add other important info...

[code]
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at f670854)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * A2plus 0.4.60n
  * ExplodedAssembly
  * GdsLightBlue 0.0.1
[/code]

kbwbe, 2023-11-05:
Hi, thank you for the patch. As i am also no expert for regular expressions, i added code which removes the &apos; html sequence.
At moment, this apos thml sequence seems to make most trouble.
the new A2p version is 0.4.61

glassondesignstudio, 2023-11-18:
That seems to be working now. Thanks.

closed 2023-11-18 by glassondesignstudio