kbwbe/A2plus #613: JSON serialisation error

glassondesignstudio, 2023-11-18:
After upgrading to 0.4.61 I found that there was a JSON serialization error occurring when I created the parts list. The error I am seeing is:

12:22:11  PropertyPythonObject::toString(): failed for <class 'a2p_importedPart_class.ImportedPartViewProviderProxy'>
12:22:11  Traceback (most recent call last):
  File "/usr/lib/python3.10/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/usr/lib/python3.10/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.10/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/usr/lib/python3.10/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable

This error doesn't seem to prevent parts lists being created but it might cause issues elsewhere. I have raised this separately because I don't think it is related to the parts list issue that I raised previously.

FreeCAD info:

[code]
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * A2plus 0.4.61
  * ExplodedAssembly
  * GdsLightBlue 0.0.1
[/code]

simao, 2024-01-09:
I am getting the same error when saving my assembly:

```
13:44:47  PropertyPythonObject::toString(): failed for <class 'a2p_importedPart_class.ImportedPartViewProviderProxy'>
13:44:47  Traceback (most recent call last):
  File "/usr/lib/python3.10/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/usr/lib/python3.10/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.10/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/usr/lib/python3.10/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
13:44:47  PropertyPythonObject::toString(): failed for <class 'a2p_importedPart_class.ImportedPartViewProviderProxy'>
13:44:47  Traceback (most recent call last):
  File "/usr/lib/python3.10/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/usr/lib/python3.10/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.10/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/usr/lib/python3.10/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
13:44:47  PropertyPythonObject::toString(): failed for <class 'a2p_importedPart_class.ImportedPartViewProviderProxy'>
13:44:47  Traceback (most recent call last):
  File "/usr/lib/python3.10/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/usr/lib/python3.10/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.10/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/usr/lib/python3.10/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
13:44:47  PropertyPythonObject::toString(): failed for <class 'a2p_importedPart_class.ImportedPartViewProviderProxy'>
13:44:47  Traceback (most recent call last):
  File "/usr/lib/python3.10/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/usr/lib/python3.10/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.10/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/usr/lib/python3.10/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable

```

Syres916, 2024-01-17:
@kbwbe I've reported the issues with 0.21.2 https://forum.freecad.org/viewtopic.php?p=732345#p732345 and the fix has gone in to the 0.21 branch (it's already in 0.22dev main branch) with preparation for 0.21.3 but there's still an issue with this workbench for Python 3.11.x and above users which I'm happy to submit a Pull Request. Do you want PR's submitted against devel branch or master?

kbwbe, 2024-01-17:
Thank you @Syres916 for help. Please submit the PR to master branch

IdahoEv, 2024-02-06:
I get this same issue despite having FreeCAD 0.21.2, A2Plus 0.4.62, and Python 3.8.   I can see that my version of a2p_importedPart_class.py on disk contains the updated code.  Am I missing something? 

[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.62
  * Assembly3 0.12.2
  * Assembly4 0.50.6
  * fasteners 0.4.63
  * FeedsAndSpeeds 0.5.0
  * Manipulator 1.5.7
  * woodworking 0.21.2.33771
[/code]

Syres916, 2024-02-07:
@IdahoEv you can either use the latest 0.22dev build of FreeCAD or downgrade to 0.21.1, the fix for the core code will be available whenever 0.21.3 is released. Unfortunately 0.21.2 was released in a bit of a rush IMHO.