kbwbe/A2plus #636: Module not found error on Arch Linux: PySide2.

noahmartinwilliams, 2024-08-30:
The exact error is: `During initialization the error "No module named 'PySide2'" occurred in /home/noah/.local/share/FreeCAD/Mod/A2plus/./InitGui.py` .
Is there a way to do it without PySide2? I've tried looking into the files and found that the one that requires it is automatically generated, and I can't find documentation on how to generate it (I'm not familiar with rcc).

Syres916, 2024-08-30:
I'm working on a local Qt6 solution for EndeavorOS and when I get a solution to the Preferences Page, I'll hopefully get a complete PR for all changes. If you just want A2plus to start then the hack is to change line 6 in _a2p_Resources3.py_ from:

`from PySide2 import QtCore`

to:

`from PySide6 import QtCore`

Two other files that I had to change were _CD_A2plusupdater.py_ and _CD_ConstraintViewer.py_ which both have a line towards the top of the files:

`from PySide import QtCore, QtGui, QtUiTools`

which I changed to:

```
from PySide import QtCore, QtGui
try:
    from PySide import QtUiTools
except ImportError:
    pass
```

Note: at present the Preferences page crashes for me so you'll need to copy the settings from another _user.cfg_ if the defaults don't work for what you need and if you update the Wb from AddonManager then all this is lost.

kbwbe, 2024-09-02:
I merged Syres916's fixes to A2plus. Please update to version 0.4.67.

noahmartinwilliams, 2026-01-20:
It seems to run now. I can't remember what exactly the problem was, but after installing it and switching to the A2Plus workbench it seems to not crash.