kbwbe/A2plus #637: <Exception> TopoShapeExpansion.cpp(5626): Null shape

kozross, 2024-08-30:
This error happens when I try to convert a Part Design part to A2Plus. It happens seemingly at random, but in [this project](https://github.com/user-attachments/files/16811200/drv8825-mount.zip), I get this consistently. The full text of the error is:

```
<Exception> TopoShapeExpansion.cpp(5626): Null shape
15:26:45  Running the Python command 'a2p_ConvertPart' failed:
Traceback (most recent call last):
  File "/home/koz/.local/share/FreeCAD/Mod/A2plus/./a2p_convertPart.py", line 181, in Activated
    convertToImportedPart(doc, s)
  File "/home/koz/.local/share/FreeCAD/Mod/A2plus/./a2p_convertPart.py", line 110, in convertToImportedPart
    topoMapper.createTopoNames(desiredShapeLabel = obj.Label)
  File "/home/koz/.local/share/FreeCAD/Mod/A2plus/./a2p_topomapper.py", line 647, in createTopoNames
    shell = Part.makeShell(faces)
            ^^^^^^^^^^^^^^^^^^^^^

FreeCAD exception thrown (Null shape)
```

Steps to reproduce:

1. Extract and open the linked project
2. Select the 'base' part
3. Click 'Convert a part to A2Plus'
4. Watch the error appear

Restarting FreeCAD doesn't solve this, and what's even weirder is that the part renders fine and doesn't have any problems in and of itself. 

I am running FreeCAD 0.22.0dev, and my A2Plus version is 0.4.66.

Syres916, 2024-08-30:
If this works using FreeCAD 0.21.x then it's a regression in FreeCAD 0.22dev which seeing as the Toponaming project hasn't completed (no matter what you may have read on other internet sites) then it's a regression in the code code and nothing that @kbwbe should have to do any serious rewriting to handle. The author of the Fasteners and Sheetmetal workbenches has already raised an issue for simple operations broke by TNP and it's being actively worked on.

kozross, 2024-09-25:
@Syres916 - in light of FreeCAD now being in its _second_ release candidate based on 0.22dev, does this make this issue current?

Syres916, 2024-09-28:
@kozross well, I did specifically ask if it works using 0.21.x and as you didn't answer I tested it myself and it doesn't so it's not a regression. The root cause of your issue is an educational one IMHO. The base object should have been created in an external file which you then use `Add a part from an external file`. If you need to change the _base.FCStd_, that's fine, make the change, save the file and then in the assembly file `Update parts imported into the assembly` I've attached the zip which includes both files to be kept in the same folder and tested both 0.21.x and

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38843 (Git)
Build type: Release
Branch: main
Hash: d519f24df9931e4a98d19ee39cdcd6124ae6a7bd
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.67
```
which has fixes in it which will go into the next RC or Release.
[drv8825-assy.zip](https://github.com/user-attachments/files/17172892/drv8825-assy.zip)

luzpaz, 2024-10-19:
Close ticket?

kbwbe, 2024-10-22:
Done.

closed 2024-10-22 by kbwbe