kbwbe/A2plus #644: Is A2plus addon ready for 1.0 release?

luzpaz, 2024-09-06:
https://blog.freecad.org/2024/08/30/1-0rc1-is-coming-next-week-lets-update-addons/

e-dub, 2024-09-12:
Getting a segmentation fault error when starting FreeCAD v 1.0.0 RC1 with A2Plus v4.67 in Ubuntu 24.04.

```
~$ FreeCAD
FreeCAD 1.0.0, Libs: 1.0.0RC1R38702 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

FEMbyGEN Workbench loaded
Sheet Metal workbench loaded
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7778ace45320]
#1  /lib/x86_64-linux-gnu/libshiboken2.cpython-312-x86_64-linux-gnu.so.5.15(SbkObjectType_SetPropertyStrings+0x13) [0x7778afbbaaf3]
#2  /usr/local/lib/python3.12/dist-packages/PySide6/QtCore.abi3.so(+0x15b176) [0x77787195b176]
#3  /usr/local/lib/python3.12/dist-packages/PySide6/QtCore.abi3.so(PyInit_QtCore+0x62) [0x7778718fb03e]
#4  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x2d641f) [0x7778aeed641f]
#5  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x2d33aa) [0x7778aeed33aa]
#6  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x1dc765) [0x7778aeddc765]
#7  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x8ec) [0x7778aed19dec]
#8  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x182d6e) [0x7778aed82d6e]
#9  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyObject_CallMethodObjArgs+0x101) [0x7778aed82fb1]
#10  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyImport_ImportModuleLevelObject+0x2e2) [0x7778aeed4a12]
#11  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x296507) [0x7778aee96507]
#12  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x1dc4ac) [0x7778aeddc4ac]
#13  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x8ec) [0x7778aed19dec]
#14  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x182d6e) [0x7778aed82d6e]
#15  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyObject_CallMethodObjArgs+0x101) [0x7778aed82fb1]
#16  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyImport_ImportModuleLevelObject+0x164) [0x7778aeed4894]
#17  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x296507) [0x7778aee96507]
#18  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x1dc4ac) [0x7778aeddc4ac]
#19  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x8ec) [0x7778aed19dec]
#20  /usr/local/lib/python3.12/dist-packages/shiboken6/libshiboken6.abi3.so.6.6(+0x12943) [0x77788ec2e943]
#21  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x1dbe88) [0x7778aeddbe88]
#22  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyObject_MakeTpCall+0x8f) [0x7778aed827af]
#23  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0xa5cc) [0x7778aed23acc]
#24  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyEval_EvalCode+0x20f) [0x7778aee9d1ff]
#25  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x299180) [0x7778aee99180]
#26  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x1dc4ac) [0x7778aeddc4ac]
#27  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x8ec) [0x7778aed19dec]
#28  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x182d6e) [0x7778aed82d6e]
#29  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyObject_CallMethodObjArgs+0x101) [0x7778aed82fb1]
#30  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyImport_ImportModuleLevelObject+0x2e2) [0x7778aeed4a12]
#31  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x296507) [0x7778aee96507]
#32  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x1dc4ac) [0x7778aeddc4ac]
#33  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x8ec) [0x7778aed19dec]
#34  /usr/local/lib/python3.12/dist-packages/shiboken6/libshiboken6.abi3.so.6.6(+0x12943) [0x77788ec2e943]
#35  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x1dbe88) [0x7778aeddbe88]
#36  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyObject_MakeTpCall+0x8f) [0x7778aed827af]
#37  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0xa5cc) [0x7778aed23acc]
#38  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyEval_EvalCode+0x20f) [0x7778aee9d1ff]
#39  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x299180) [0x7778aee99180]
#40  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x66d3) [0x7778aed1fbd3]
#41  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyEval_EvalCode+0x20f) [0x7778aee9d1ff]
#42  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x2f7af8) [0x7778aeef7af8]
#43  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x2f7c1c) [0x7778aeef7c1c]
#44  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyRun_StringFlags+0x74) [0x7778aeefb234]
#45  0x7778af957c88 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/local/lib/libFreeCADBase.so+0x146
#46  0x7778b252c50c in Gui::Application::runInitGuiScript() from /usr/local/lib/libFreeCADGui.so+0x50
#47  0x7778b272a7df in Gui::StartupPostProcess::showMainWindow() from /usr/local/lib/libFreeCADGui.so+0x77
#48  0x7778b2728e4a in Gui::StartupPostProcess::execute() from /usr/local/lib/libFreeCADGui.so+0xa0
#49  0x7778b252d525 in Gui::Application::runApplication() from /usr/local/lib/libFreeCADGui.so+0x185
#50  FreeCAD(+0x26c6e) [0x5dd60111ec6e]
#51  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7778ace2a1ca]
#52  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7778ace2a28b]
#53  FreeCAD(+0x25da5) [0x5dd60111dda5]
```

kbwbe, 2024-09-12:
Is working fine here with recent weekly build:

OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git) AppImage
Build type: Release
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: //
Installed mods: 
  * A2plus 0.4.67

So i cannot reproduce the reported seg fault mentioned above. Does someone else have same problems?

luzpaz, 2024-09-12:
@kbwbe can you test on a Qt6 build ?

kbwbe, 2024-09-13:
Where can i find a suitable Appimage build with Qt6 for Linux ?
Weekly builds seem to be delivered using Qt5.

luzpaz, 2024-09-13:
@adrianinsaval how far are we from having this as an option for Appimages ?

adrianinsaval, 2024-09-14:
qt6 will be primarily used after 1.0, for 1.0 we are primarily targeting qt5

luzpaz, 2024-09-18:
Thanks @kbwbe for testing.  
Thanks @adrianinsaval for the feedback.

closed 2024-09-18 by luzpaz

adrianinsaval, 2024-09-18:
There might be a qt6 flatpak soon or if you are feeling adventurous archlinux is shipping RC1 built with qt6 in their repos. Assuming you're not running arch and don't want to distro hop just to test that you could try running arch with distrobox