kbwbe/A2plus #651: Error importing numpy

asafecosta, 2025-04-06:
Hi!

I'm on Mac OS Sequoia 15.4. The A2plus suddenly stopped work. I have reinstalled many times and this error continues appearing.

16:39:49  During initialization the error "Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there." occurred in /Users/[user]/Library/Application Support/FreeCAD/Mod/A2plus/./InitGui.py
16:39:49  Please look into the log file for further information

kbwbe, 2025-04-07:
Hi,
some questions:
1) which version of A2plus do you use ?
2) is there more information output except this one 
"occurred in /Users/[user]/Library/Application Support/FreeCAD/Mod/A2plus/./InitGui.py"

asafecosta, 2025-04-08:
Hi!
 1. I am using the 0.4.68 version.

2. I have tried to catch the log of initialization.

07:53:11  OpenGL version is: 2.1 (2.1 Metal - 89.4)
07:53:12  Run Gui init script
07:53:12  Init: Running FreeCADGuiInit.py start script...
07:53:12  Init:   Searching modules...
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Mesh... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Draft... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/ReverseEngineering... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Plot(InitGui.py not found)... ignore
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Test... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Fem... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Spreadsheet... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Web(InitGui.py not found)... ignore
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/TechDraw... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Assembly... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/MeshPart... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Show(InitGui.py not found)... ignore
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Sketcher... done
07:53:12  Loading GUI of Start module... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Start... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/OpenSCAD... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Surface... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Robot... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Inspection... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Idf(InitGui.py not found)... ignore
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/CAM... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/AddonManager... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Points... done
07:53:12  Loading Material module... done
07:53:12  Module: Part
07:53:12  Loading Part module... done
07:53:12  Loading Measure module... done
07:53:12  Loading GUI of Measure module... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Measure... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/BIM... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/PartDesign... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Tux... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Part... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Import... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Material... done
07:53:12  Init:      Initializing /Applications/FreeCAD.app/Contents/Resources/Mod/Help... done
07:53:12  Init:      Initializing /Users/asafecostalopes/Library/Application Support/FreeCAD/Mod/A2plus/./... failed
07:53:12  ----------------------------------------------------------------------------------------------------
07:53:12  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/numpy/_core/__init__.py", line 23, in <module>
    from . import multiarray
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/numpy/_core/multiarray.py", line 10, in <module>
    from . import overrides
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/numpy/_core/overrides.py", line 8, in <module>
    from numpy._core._multiarray_umath import (
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/numpy/_core/_multiarray_umath.cpython-311-darwin.so, 0x0002): Library not loaded: @rpath/libgfortran.5.dylib
  Referenced from: <C435405B-BFF1-399F-A1EC-BD43418F546A> /Applications/FreeCAD.app/Contents/Resources/lib/libopenblas.0.dylib
[... cut ...]
07:53:13  Wizard shaft module cannot be loaded
07:53:13  Loading GUI of Material module... done
07:53:13  Loading GUI of Part module... done
07:53:13  Loading GUI of Sketcher module... done
07:53:13  Loading GUI of PartDesign module... done
07:53:13  Init: Showing main window
07:53:13  Main window restored
07:53:13  Show main window
07:53:13  Toolbars restored
07:53:13  3Dconnexion framework not found!
07:53:13  Init: Entering event loop
07:53:13  Init: Executing event loop...
07:53:13  Init: Processing command line files

kbwbe, 2025-04-08:
Hi,
thank you for the error dump. The error seems to deal with library installation problems as there are such messages as:

Reason: tried: '/Applications/FreeCAD.app/Contents/Resources/lib/libgfortran.5.dylib' (duplicate LC_RPATH '@loader_path')

There is a hint that there is a duplicate path somehow which is bad in generally. Also the numpy installation seems to be conflicting with  shiboken because of this part of messages:

File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/numpy/_core/multiarray.py", line 10, in
from . import overrides
File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
return original_import(name, *args, **kwargs)

As A2plus is using numpy, it needs a correct installation of this module. I guess, the error is produced by something outside of A2plus. Maybe a recent software update or something within FreeCAD itself.

Did you do some software updates just before A2plus errors occured?
And what FC version are you using ?

asafecosta, 2025-04-09:
Yes, I had a System update last week. Apparently this caused the problem. I also tried reinstall/update python, numpy by conda and brew both, I saw a hint to change the path that execute python in InigGui.py of a2plus file. But all these trials had no effect to solve the problem. =/

asafecosta, 2025-04-17:
Thank for the advices! unfortunately it didn't worked. 

I'll continue trying, if I have success I'll come back to share the information.

asafecosta, 2025-04-23:
Hi! returning to share the solution.
I have found today in this FreeCAD forum, and it worked for me!

[Forum](https://forum.freecad.org/viewtopic.php?style=8&t=96065)

closed 2025-04-23 by asafecosta